pybricks/pybricks-micropython #2: UART hotplug is broken

dlech, 2020-06-23:
https://github.com/pybricks/support/issues/43

Currently UART devices can only be detected once. If it is unplugged or goes out of sync, then the hub has to be rebooted.

We need to add some events to communicate between the uartdev and the lpf2_port.

laurensvalk, 2020-12-08:
This seems to be working fine on the SPIKE Prime Hub. It is still confirmed on Move Hub. (Have not tried the others.)

So if it is partially there already, maybe we don't need a significant overhaul to `uartdev` to fix this after all.

laurensvalk, 2020-12-08:
Update: This only works if the new device is also a UART device. Hotplugging passive <---> passive is also OK, but passive <---> UART is not. Still, just making hotplugging UART/UART work on all hubs is probably still worth it.

laurensvalk, 2020-12-09:
UART to UART hotplugging is fixed :tada:.

I'll suggest a patch for UART to passive hotplugging also later today.

laurensvalk, 2020-12-09:
Patch for passive sensors is pushed also. Pbio tests fail since there isn't a test for `ioport_lpf2` at the moment.

laurensvalk, 2020-12-09:
@dlech , would you like to review these changes?

dlech, 2020-12-09:
I made a PR for easier review.

laurensvalk, 2020-12-10:
A couple of things turned up that I'd like to address also.

- [ ] On Technic Hub, sensors always fail at first, then succeed on the second sync. The easiest way to see this is with the ColorDistanceSensor. I think the error is always `Bad device type id`, so we should be able to deterministically figure out why, and then fix it.
- [ ] I'm currently resetting the ioport to check for passive sensors after every error in `uartdev.c` which is obviously not needed. It would be better to do this only on disconnect or persistent failure to sync up.
- [ ] Avoid using "processes.h" --- see @dlech's comments in the PR. As a first step, I've dropped the commits that messed with the tests and the debug port, instead wrapping the call to the process in an `#if` block for now. This too will be removed and replaced by a call to the right namespace instead.

laurensvalk, 2020-12-11:
I was looking at fixing the baudrate on syncup, but this requires more changes than I thought, so I think we can defer that to the more rigorous revisions after this release. For now it's probably better if the sensors just work, even if people have to wait a few seconds.

laurensvalk, 2020-12-12:
Making some good progress here, but new things tend to pop up as we go along

- [ ] Do not auto-power off all power ports (we do this in the motor drivers) because it turns off all sensors too
- [ ] wait for sync to complete before returning enodev

We've got a few minor motor patches pending so it would be nice to merge those first to avoid conflicts before fixing the first one here.

commit 09031456 (2020-12-09): pbdrv/ioport: run dcm if uart sensor unplugged

This fixes hotplugging of passive sensors after a UART sensor has been unplugged.

Combined with the prevous commits, this fixes #2.
commit 76437ccf (2020-12-09): pbdrv/ioport: run dcm if uart sensor unplugged

This fixes hotplugging of passive sensors after a UART sensor has been unplugged.

Combined with the prevous commits, this fixes #2.
commit 08401fa2 (2020-12-09): pbdrv/ioport: run dcm if uart sensor unplugged

This fixes hotplugging of passive sensors after a UART sensor has been unplugged.

Combined with the prevous commits, this fixes #2.
commit e24fe096 (2020-12-09): pbdrv/ioport: run dcm if uart sensor unplugged

This fixes hotplugging of passive sensors after a UART sensor has been unplugged.

Combined with the prevous commits, this fixes #2.
closed 2020-12-23 by laurensvalk