pybricks/pybricks-micropython #3: [Question]

GianCann, 2020-06-25:
I'm trying to study in depth the operation of the firmware because I'm curious to understand also the hardware part.
Reading the file https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/platform/cplus_hub/platform.c I would like to ask why PC12 and PB12 are put on high level.

    // Keep main power on (PC12)
    gpio_init.Pin = GPIO_PIN_12;
    gpio_init.Mode = GPIO_MODE_OUTPUT_PP;
    HAL_GPIO_WritePin(GPIOC, GPIO_PIN_12, GPIO_PIN_SET);
    HAL_GPIO_Init(GPIOC, &gpio_init);

    // Turn VCC_PORT on (PB12)
    HAL_GPIO_WritePin(GPIOB, GPIO_PIN_12, GPIO_PIN_SET);
    HAL_GPIO_Init(GPIOB, &gpio_init);

dlech, 2020-06-25:
Disclaimer: Everything in the firmware has been done through reverse engineering, so everything is just a best guess.

PC12 is the main power. It has to be held high to keep the power on. It is set low at shutdown to power off the hub.

PB12 is the power for pin 4 on the I/O ports (and possibly other things like the encoders of the built-in motors on Move hub). We just turn it on and keep it on. But according to the LEGO Wireless Protocol, their firmware can turn this on and off at runtime. See "VCC Port control on/off" at https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#action-types.

GianCann, 2020-06-27:
> PB12 is the power for pin 4 on the I/O ports (...) See "VCC Port control on/off"

Ok, thank you for this clarification. So, PB12 manage (on/off) on ALL ports of the hubs.
I thought it was possible to disable / enable it for single port. Good to know.