pybricks/pybricks-micropython #4: [Question] Normal Boot & Bootloader: How it works?

GianCann, 2020-06-27:
I'm continuing to study the Pybricks firmware and now I'm trying to understand the hub startup procedure.

(https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/platform/cplus_hub/sys.c)

```
// Bootloader reads this address to know if firmware loader should run
uint32_t bootloader_magic_addr __attribute__((section(".magic")));
#define BOOTLOADER_MAGIC_VALUE  0xAAAAAAAA
(...)
void pbsys_reboot(bool fw_update) {
    if (fw_update) {
        bootloader_magic_addr = BOOTLOADER_MAGIC_VALUE;
    }
    // this function never returns
    NVIC_SystemReset();
}
```

My questions are:
The bootloader to flash a new firmware, is a (LEGO) firmware in another space of memory?

Where is it decided which of the two firmware (bootloader or pybricks) should be loaded?

dlech, 2020-06-27:
Again, this is just what I have learned from reverse engineering, so not guaranteed to 100% accurate...

The STM32 chips used in the Powered Up hubs start by reading the stack pointer memory location, the reset vector and the interrupt vector table from the beginning of flash memory, then program control jumps to the address given by the reset vector.

At the factory, LEGO has programmed a "bootloader" (LWP3 docs call this Boot Loader service or Flash Loader). This area of the flash memory is write protected, so it is theoretically impossible to accidentally wipe out the bootloader and brick your brick. The size of this flash memory area can be seen in the linker (*.ld) scripts.

When the bootloader first starts, it checks to see if the button is being held down or if a certain RAM address contains a certain value (the ones you have pointed out). If either one is true, then it starts the BLE bootloader service for flashing the firmware, otherwise it sends program control to the firmware that was flashed. The main entry point is `Reset_Handler` in `startup.s` which initializes the RAM (global variable initial values), calls the `SystemInit` function in `platform.c` and finally the MicroPython `main()` function.

GianCann, 2020-06-27:
Thanks @dlech ,
you're always very kind.
I do not have high knowledge of the functioning of microcontrollers and I am trying to learn also thanks to your interesting reverse engineering work.