pybricks/pybricks-micropython #5: Battery tasks

dlech, 2020-06-27:
We already have basic battery current and voltage monitoring and shutdown on low battery. Some outstanding tasks:

- [ ] Confirm accuracy of current and voltage readings.
- [ ] Use hub LED to indicate low battery.
- [ ] Allow for different battery chemistries.
- [ ] Refactor battery monitoring out of `sys.c` and share more code between platforms.
- [ ] Provide battery indication to pybricks-code IDE.

Current battery levels are defined as follows:

```C
#define BATTERY_OK_MV           6000    // 1.0V per cell
#define BATTERY_LOW_MV          5400    // 0.9V per cell
#define BATTERY_CRITICAL_MV     4800    // 0.8V per cell
```

dlech, 2020-06-27:
For reference, here are the battery levels used in EV3. `BATT` is 6 Alkaline cells and `ACCU` is the rechargeable Li-on battery. 

[source](https://github.com/mindboards/ev3sources/blob/78ebaf5b6f8fe31cc17aa5dce0f8e4916a4fc072/lms2012/lms2012/source/lms2012.h#L433-L472)

```C
#define   BATT_INDICATOR_HIGH           7500          //!< Battery indicator high [mV]
#define   BATT_INDICATOR_LOW            6200          //!< Battery indicator low [mV]

#define   ACCU_INDICATOR_HIGH           7500          //!< Rechargeable battery indicator high [mV]
#define   ACCU_INDICATOR_LOW            7100          //!< Rechargeable battery indicator low [mV]

#define   LOW_VOLTAGE_SHUTDOWN_TIME     10000         //!< Time from shutdown lower limit to shutdown [mS]

#define   BATT_WARNING_HIGH             6.2           //!< Battery voltage warning upper limit [V]
#define   BATT_WARNING_LOW              5.5           //!< Battery voltage warning lower limit [V]
#define   BATT_SHUTDOWN_HIGH            5.5           //!< Battery voltage shutdown upper limit [V]
#define   BATT_SHUTDOWN_LOW             4.5           //!< Battery voltage shutdown lower limit [V]

#define   ACCU_WARNING_HIGH             7.1           //!< Rechargeable battery voltage warning upper limit [V]
#define   ACCU_WARNING_LOW              6.5           //!< Rechargeable battery voltage warning lower limit [V]
#define   ACCU_SHUTDOWN_HIGH            6.5           //!< Rechargeable battery voltage shutdown upper limit [V]
#define   ACCU_SHUTDOWN_LOW             6.0           //!< Rechargeable battery voltage shutdown lower limit [V]
```

@ZPhilo do the `ACCU` values sound reasonable for the Li-on battery pack you have made?

ZPhilo, 2020-06-28:
Hi David,
The discharge curve here tells most of the story
https://philohome.com/pupbat/CPDischarge.gif
If I understand well, a "battery running low" warning is issued between ACCU_WARNING_HIGH and ACCU_WARNING_LOW, and a shutdown is done when voltage is between ACCU_SHUTDOWN_HIGH and ACCU_SHUTDOWN_LOW? 
I don't quite understand the need for ACCU_SHUTDOWN_LOW since hopefully the shutdown will have occured above that voltage?

dlech, 2020-06-28:
If I am remembering correctly, the official firmware for EV3 will turn on the low battery light when the voltage drops below ACCU_WARNING_LOW and will turn the light back off it the voltage rises back above ACCU_WARNING_LOW. Likewise, the brick will shut down if the voltage drops below ACCU_SHUTDOWN_LOW for 10 seconds and will cancel the shutdown if the voltage rises back above ACCU_SHUTDOWN_HIGH.

I saw a note on [this page](https://philohome.com/batteries/discharge-nxt.gif) that the EV3 rechargeable battery runs at a bit lower voltage compare to the other Li-ion batteries so that might explain the difference.

ZPhilo, 2020-06-28:
It's quite possible that some protection circuitry reduces a bit the voltage output of LEGO batteries. I think that the voltages you quote are nonetheless OK for "naked" LiPo batteries like I did.

laurensvalk, 2020-10-09:
I've expanded the task list with enabling battery charging on SPIKE Prime, which we currently don't have yet.

closed 2021-03-29 by dlech