pybricks/pybricks-micropython #6: Compiling firmware on Windows don't work

GianCann, 2020-06-28:
I easily compiled the firmware on Linux, but was unable to do it on Windows.
I installed the ARM toolchain, MinGW:

```
C:\Users\remoteadmin\pybricks-micropython>c:\mingw\bin\make.exe V=1 cplushub
process_begin: CreateProcess(NULL, uname -s, ...) failed.
make[1]: Entering directory 'C:/Users/remoteadmin/pybricks-micropython/micropyth
on/mpy-cross'
mkdir -p build/genhdr
Sintassi del comando errata.
../py/mkrules.mk:121: recipe for target 'build/genhdr' failed
make[1]: *** [build/genhdr] Error 1
make[1]: Leaving directory 'C:/Users/remoteadmin/pybricks-micropython/micropytho
n/mpy-cross'
Makefile:84: recipe for target 'mpy-cross' failed
make: *** [mpy-cross] Error 2

C:\Users\remoteadmin\pybricks-micropython>
```

laurensvalk, 2020-06-28:
Can you try cloning upstream micropython (instead of our fork), go into the mpy-cross folder and run make?

GianCann, 2020-06-28:
Ok, I've tried: it's a Micropython problem...

```
C:\Users\remoteadmin\micropython>cd mpy-cross

C:\Users\remoteadmin\micropython\mpy-cross>make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosit
y.
process_begin: CreateProcess(NULL, uname -s, ...) failed.
mkdir -p build/genhdr
Sintassi del comando errata.
../py/mkrules.mk:121: recipe for target 'build/genhdr' failed
make: *** [build/genhdr] Error 1

C:\Users\remoteadmin\micropython\mpy-cross>
```

dlech, 2020-06-28:
I would recommend using the Windows Subsystem for Linux instead.

GianCann, 2020-06-29:
Thank you @dlech ! With WSL all is perfect on Windows 10.
Saturday I have tried on a "old" Windows 7 PC that not support WSL.

closed 2020-06-29 by GianCann