pybricks/pybricks-micropython #7: City Hub Port A polarity is flipped

laurensvalk, 2020-06-29:
```python
from pybricks.pupdevices import Motor
from pybricks.parameters import Port
from pybricks.tools import wait

motor = Motor(Port.A)
motor.dc(50) # Should go clockwise, but it goes counterclockwise.
wait(10000)
```
Naturally, this causes all controlled `run` commands to fail because of the positive feedback loop.

laurensvalk, 2020-06-29:
When we fix it, we can fix this `FIXME` for MoveHub too. It's probably broken the same way because we copied it.
https://github.com/pybricks/pybricks-micropython/blob/master/extmod/modmotor.c#L212

For Move Hub we need to invert the counter on Port A as well.

GianCann, 2020-06-29:
Aaah!!!
This explain to me why the run_angle or run_time don't stop the motor...

laurensvalk, 2020-06-29:
Fixed by https://github.com/pybricks/pybricks-micropython/commit/383786b252a6372308e90a0870d4ec6893d89b0c

You can try it using the latest successful build on master.

> When we fix it, we can fix this FIXME for MoveHub too.

Fixed by https://github.com/pybricks/pybricks-micropython/commit/cc2f9c2ce06346e9ca89aedf265c31057f7d7e46, https://github.com/pybricks/pybricks-micropython/commit/ee678d637b7efc6025a54cc67396829064894e1a, https://github.com/pybricks/pybricks-micropython/commit/5b0be4ecf4b72f2ddcb590f31b4d0dc309e1cc29.

closed 2020-06-29 by laurensvalk