pybricks/pybricks-micropython #11: Permanent program does not run if Bluetooth is not connected

laurensvalk, 2020-08-25:
Since the recent changes to the system lights and `sys/hmi`, the permanent program does not run if Bluetooth is not connected.

The permanent program is still working on SPIKE Prime since it isn't waiting for Bluetooth.

I've just updated the `main.py` scripts to just wait a few seconds so it's easier to test this workflow while we are revising the system lights and UI.

dlech, 2020-08-25:
I think the program actually does run still but just the indication light doesn't work. This is because we don't have a way to stop Bluetooth advertising yet without connecting Bluetooth first.

laurensvalk, 2020-12-11:
Next step: do program run animation similar to light matrix animation during program run for now.

closed 2020-12-11 by dlech