pybricks/pybricks-micropython #13: Light not working on PrimeHub

laurensvalk, 2020-08-27:
This no longer seems to do anything, light stays green.

```python
from pybricks.tools import wait
from pybricks.parameters import Color
from pybricks.hubs import PrimeHub

hub = PrimeHub()
hub.light.on(Color.RED)

wait(5000)
print("Hello, World!")
```

Blinking and patterns also don't work, which also applies to EV3.

dlech, 2020-08-27:
It works for me on PrimeHub built from commit a7b9ed9de2d6230d2be3c9e119de47ba7fb8b4cc. Maybe need to clean the build directory?

dlech, 2020-08-27:
EV3 light animations are fixed by a8a5de188d5568522698bd7870872fc10056e8f9.

laurensvalk, 2020-08-28:
Reproducible case has already been posted so it is not blocked. This happens just as well on a7b9ed9de2d6230d2be3c9e119de47ba7fb8b4cc, also with clean builds.

REPL is probably just a bit slower. The light works if adding extra pauses too, but that shouldn't be needed. This might be a hint of where the problem is though.