pybricks/pybricks-micropython #15: Git commit conventions for changes to pybricks package

laurensvalk, 2020-09-03:
Here's a thought to simplify commit messages for the builtin `pybricks` package while making them clearer as well. To be entirely consistent, we'd currently have to write:

```
pybricks/pupdevices/pb_type_pupdevices_pfmotor: description of change
```

In practice, the latter often gets truncated to the class name, i.e. `PFMotor`, but we could be more consistent about capitalization. Since this is the implementation of a Python package, how about this instead?

```
pybricks.pupdevices.PFMotor: description of change
```

laurensvalk, 2020-09-03:
Other than this, [this](https://github.com/pybricks/pybricks-micropython/commits/work) is ready for merging.

dlech, 2020-09-03:
This is fine with me.

https://github.com/pybricks/pybricks-micropython/blob/master/CONTRIBUTING.md#commit-messages should be updated to reflect this new style as well.

dlech, 2020-09-03:
> but we could be more consistent about capitalization

Off topic, but along these lines, I think it makes sense to use the same capitalization in the file names as well, e.g. `pb_type_pupdevices_PFMotor.c`

laurensvalk, 2020-10-09:
We are using this style now, and the note was added here. https://github.com/pybricks/pybricks-micropython/commit/e260b30a0d4d11ea5c181e8b46965f365f149352

Let's keep using lowercase filenames to avoid conflicts on systems which ignore casing.

closed 2020-10-09 by laurensvalk