pybricks/pybricks-micropython #16: Blink patterns should start with on phase

laurensvalk, 2020-09-03:
When you press the on button, it takes about a second before you see the light, even though the hub has long since finished booting. This makes the UI confusing, leading to problems like [this issue](https://github.com/pybricks/support/issues/121).

This happens because the advertising pattern is currently defined like this:

```c
    [PBSYS_STATUS_LIGHT_INDICATION_BLE_ADVERTISING] =
        (const pbsys_status_light_indication_pattern_element_t[]) {
        { .color = PBIO_COLOR_BLACK, .duration = 22 },
        { .color = PBIO_COLOR_BLUE, .duration = 1 },
        { .color = PBIO_COLOR_BLACK, .duration = 2 },
        { .color = PBIO_COLOR_BLUE, .duration = 1 },
        PBSYS_STATUS_LIGHT_INDICATION_PATTERN_END
```

This was fixed for [user patterns](https://github.com/pybricks/pybricks-micropython/commit/5c7a76cf2f4022737e3967ff5d2d5556cfe09ff7) but not yet in system patterns.

It can be fixed by putting the off period at the end, but we may need to update other patterns also to prevent conflicts.

dlech, 2020-09-03:
I don't think this can be fixed in our firmware. The bootloader does not transfer control to our firmware until the button is released. The bootloader also already sets the light to solid blue when it transfers control so changing the order would just make the light blue slightly longer at first boot.

laurensvalk, 2020-09-03:
Just to make sure we're talking about the same thing here:

> The bootloader does not transfer control to our firmware until the button is released.

The light stays off for about a second *after* releasing the button.

Before we had the advertising pattern, we made the light blue immediately after releasing, so it seems to me we do have control over the the light at that point.

dlech, 2020-09-03:
> The light stays off for about a second _after_ releasing the button.

It looks like movehub is different - it has the light off when the button is released. I checked cityhub and cplushub previously and both are solid blue at boot.

closed 2020-09-04 by dlech