pybricks/pybricks-micropython #18: Coveralls - PBIO — Coverage decreased (-0.0%)

laurensvalk, 2020-10-01:
Recent builds appear to fail with ❌, even though it passed on a branch earlier and no changes to pbio were made.

Example: https://github.com/pybricks/pybricks-micropython/commit/b257e4a9a00b0fed170cdff70f0325054943868a

dlech, 2020-10-02:
It appears that the motor control test (or possibly some other test - but motorpoll.c is listed as one of the changed files) is nondeterministic. The thing that changed that caused the failure was how many times certain lines of code ran. More specifically, the tests completed with the contiki event loop being polled fewer times. Off hand, I'm not really sure how this could happen since the motor test is not using a wall clock for measuring time but rather has full control over the clock value during the test.

laurensvalk, 2020-10-28:
How about turning this check off until we fix it?

dlech, 2020-11-09:
finally got a :heavy_check_mark:

laurensvalk, 2020-11-10:
:heavy_check_mark: :heavy_check_mark: :heavy_check_mark: :heavy_check_mark:

laurensvalk, 2020-11-10:
https://github.com/pybricks/pybricks-micropython/commit/34675354369d48bac9b285acc6df72f297473acc ❌

dlech, 2020-11-12:
:heavy_check_mark:

laurensvalk, 2020-11-12:
:tada:

dlech, 2020-11-13:
Not sure what is up with this one: https://coveralls.io/jobs/70727933

dlech, 2020-11-13:
Looks like we are not the only ones with this problem: https://github.com/lemurheavy/coveralls-public/issues/1446

dlech, 2020-11-13:
I found a setting that allows setting the coverage change threshold for failure. I set it to `1`, so we will see if that helps.

dlech, 2020-11-20:
Seems to be fixed now.

closed 2020-11-20 by dlech