pybricks/pybricks-micropython #19: Configuring RAM for SPIKE Prime

laurensvalk, 2020-10-09:
We currently don't use all the available RAM on this hub. We have previously considered fixed sizes for stack and heap on the two RAM segments, but we could also pick something else if that doesn't make too much of a difference.

This issue is to:
- [ ] Fix the linker script
- [ ] Fix `PYBRICKS_HEAP_KB` accordingly, potentially also reconsidering it for the other hubs while we do so.

dlech, 2020-10-09:
The linker script for SPIKE Prime is already using all RAM (320K). SRAM1 and SRAM2 can be treated as a single contiguous memory area.

The linker script for Technic Hub is the one that has the RAM split into two banks. So the simplest thing to do would be to treat it as one continuous memory like other hubs.

We can also look at replacing `PYBRICKS_HEAP_KB` with the way upstream MicroPython uses linker scripts to tune heap and stack sizes.

laurensvalk, 2020-10-09:
One way or the other, `PYBRICKS_HEAP_KB = 64` isn't nearly using the RAM it could get, which is what the original post was about. Also, the primehub linker script seems to define both `_heap_start` and `_heap_end` as well, seemingly with a different size.

> We can also look at replacing PYBRICKS_HEAP_KB with the way upstream MicroPython uses linker scripts to tune heap and stack sizes.

I don't have a particular preference. A fixed size `PYBRICKS_HEAP_KB` isn't so bad as it is easier to communicate to users. It avoids unexpected decreases as updates come out. But 64K is just a little bit too much on the safe side when it could easily be 256K :)

dlech, 2020-10-10:
Fixed by:

- Prime hub heap size: f759d15a9659c68bcc9edaafdbebd3f57039dbfe
- Technic hub RAM banks: 6b48ad4aefaaed02caa90748e891db6965b594d8
- Technic hub heap size: d09ca3af9dce626420329af52b44b3be311144cb

laurensvalk, 2020-10-11:
Excellent!

closed 2020-10-11 by laurensvalk