pybricks/pybricks-micropython #23: FW_VERSION on CI

laurensvalk, 2020-10-12:
```
FW_VERSION := $(shell $(GIT) describe --tags --dirty --always)
```

Building locally, this yields

```
@pybricks/firmware/v3.0.0+firmware.v3.0.0a9-19-g14004fb1
```

but on the CI it is only this:
```
14004fb
```

Is there a way to get the info string `v3.0.0a9-19-g14004fb1` on CI as well? This allows us to add the version to the dual boot script.

dlech, 2020-10-12:
We need to add `--exclude @pybricks/*` to the `git describe` so that even locally it will filter out the npm package tags.

Then on CI, we need to figure out how to fetch tags, preferably without fetching the entire git history.

Or just go for the easy solution: https://github.com/actions/checkout#fetch-all-history-for-all-tags-and-branches

dlech, 2020-10-12:
> Then on CI, we need to figure out how to fetch tags, preferably without fetching the entire git history.

https://github.com/actions/checkout/issues/338

dlech, 2020-10-12:
> Then on CI, we need to figure out how to fetch tags, preferably without fetching the entire git history.

This looks like it could do the trick.

https://github.com/actions/checkout/issues/217#issuecomment-636481619

dlech, 2020-10-12:
Fixed by 99b79936dc53716db6c66b00e3dfabfe7fc959d4

(we have a pretty lean git history, so full checkout is not as bad as I expected :smile: )