pybricks/pybricks-micropython #29: Firmware v3.0.0a11 broke something

JorgePe, 2020-11-21:
Sorry, I know this is not a proper bug report but I cannot send the whole script I am currently using:

I am using the Droid Commander elements for a robot MOC - Move Hub, the 2 internal motors (A and B) and the external motor (port C) and the ColorDistanceSensor (port D). The robot detects one of 6 custom colors (pre-set with detectable_color() ) and moves motors according to the color.
Also using getchar() so I can order same actions from my laptop and print() to show the detected color or 'Unsure' when not.

I had a script running OK with 3.0.0.a10
After flashing a11 I sent an action that just uses motor in port B
```
for i in range(0, 5):
     rightFoot.run_angle(900, 120  then=Stop.COAST, wait=True)
     rightFoot.run_angle(900, -120, then=Stop.COAST, wait=True)
```
It run much faster and never ended. Also couldn't force the Hub to stop or power off, had to remove batteries.

Then I tried just an action just with the color detection - it fail to detect and crashed several times.
Also got an error after entering the main loop:

```
Traceback (most recent call last):
  File "main.py", in <module>
OSError: [Errno 19] ENODEV
```
I am going to try to reduce my script to the essential so I can reproduce the error

laurensvalk, 2020-11-21:
Thanks for reporting @JorgePe!

This was broken by 9b84f4f79e23e293ef9953322fa659c6fce8e652

laurensvalk, 2020-11-21:
A shorter example to reproduce is

```python
from pybricks.pupdevices import Motor
from pybricks.parameters import Port

motor = Motor(Port.B)

motor.run_angle(900, 120)
motor.run_angle(900, -120)
```

laurensvalk, 2020-11-21:
I'm not sure how to make sense of this yet. I've been trying to break down 9b84f4f until this no longer happens. My batteries died when I reached 715fcd8, so I'll keep trying. But I'm starting to doubt that this is truly the commit that caused it, or that we're seeing something else. In any case, it seems unrelated to the PWM change.

dlech, 2020-11-21:
I think there are multiple problems. The motors still don't run correctly before commit 9b84f4f. Commit 9b84f4f seems to only introduce the hub shutting off part of the issue.

laurensvalk, 2020-11-22:
It could be worth checking if temporarily disabling auto-shown on low battery makes any difference. Then we can have a look in the motor logs to see if anything unusual happened.

laurensvalk, 2020-11-24:
This one line reproduces it for that commit: https://github.com/pybricks/pybricks-micropython/commit/47e67ba8f94fc09ec58ca00e5671276c8cf7cee0#diff-36a9b4c1bd5a5882c0c831198fc00e605390a13eb204b3f55e716699ce52a3dfR77 (The main.py added here is not the problem, that's just there to reduce the pain of testing all the commits without connecting to Pybricks Code every time)

So I wonder if we're just getting close to allocating ever so slightly too much static variables on Move Hub.

laurensvalk, 2020-11-24:
Confirmed as a memory error, unrelated to motors. For now, I've reduced the available memory for user code, but we may need to do some proper optimizations instead.

@JorgePe, do you want to test if this fixes it, by manually installing the firmware? You can do that using the permament program installation method.

closed 2020-11-24 by laurensvalk

JorgePe, 2020-11-24:
It fixes it!
Whole original script working again (except for the hub light blinking while motors working... probably batteries need to be replaced after so much wear)

Mem Info (with my script):
stack: 3758080444
GC: total: 6016, used: 3296, free: 2720
 No. of 1-blocks: 36, 2-blocks: 4, max blk sz: 59, max free sz: 106

commit 91ba6cc6 (2020-11-24): bricks/movehub: reduce heap size

We were starting to run out of RAM for stack and statically allocated variables on Move Hub, which has only 16K of RAM.

This works around it by reducing the heap for the MicroPython user program.

This fixes #29, but only temporarily. We may need to disable some features on Move Hub instead or optimize some of the drivers, such as the motor controllers.