pybricks/pybricks-micropython #30: BTstack compilation conflict when MicroPython fully checked out

laurensvalk, 2020-12-02:
A clean checkout succeeds:

```
git clone https://github.com/pybricks/pybricks-micropython.git bttest
cd bttest
make primehub  # This works
```

But it fails if MicroPython's submodules are checked out or if you cloned Pybricks recursively:

```
cd micropython
git submodule update --checkout --init
cd ..
make primehub  # Now it fails
```

It seems to be trying to compile MicroPython's btstack instead of our btstack submodule:

```
CC ../../micropython/lib/btstack/src/ad_parser.c
CC ../../micropython/lib/btstack/src/btstack_audio.c
CC ../../micropython/lib/btstack/src/btstack_base64_decoder.c
CC ../../micropython/lib/btstack/src/btstack_crypto.c
In file included from ../../lib/btstack/src/hci.h:56:0,
                 from ../../micropython/lib/btstack/src/ble/gatt_client.h:42,
                 from ../../micropython/lib/btstack/src/btstack_event.h:58,
                 from ../../micropython/lib/btstack/src/btstack_crypto.c:44:
../../lib/btstack/src/gap.h:138:1: error: unknown type name 'hci_role_t'
 hci_role_t gap_get_role(hci_con_handle_t connection_handle);
 ^~~~~~~~~~
```

laurensvalk, 2020-12-02:
The workaround is

```
cd micropython
git submodule deinit lib/btstack/
cd ..
make clean-primehub
make primehub
```

dlech, 2020-12-02:
MicroPython has tons of submodules we don't use so best practice is to just let the makefiles checkout the required submodules automatically rather than doing anything with submodules manually.

I guess the conflict is because we put btstack in the exact same directory structure `lib/btstack` and the MicroPython makefiles search both the top level directory and the `micropython` directory for all relative paths.

So we can either add something to the Pybricks stm32 makefile to deinit the MicroPython btstack submodule the same we we init other submodules or we can change the path somehow (e.g. rename `lib` to something else).

laurensvalk, 2020-12-02:
Manually undoing it as above is probably good enough, so I think we can close this.

closed 2020-12-02 by laurensvalk