pybricks/pybricks-micropython #36: Turn off power on shutdown on Prime Hub if USB is connected

laurensvalk, 2020-12-12:
On Prime Hub, the following devices stay on after auto-shutdown if USB is still connected:

- Light matrix
- Hub status light
- Sensor logic voltage
- Sensor / Motor power terminals

_Originally posted by @laurensvalk in https://github.com/pybricks/support/issues/79#issuecomment-743728863_

laurensvalk, 2020-12-14:
This also happens for other hubs as long as you are holding down the button.

dlech, 2021-06-30:
I think this is done enough to close. Battery charging and NXT support are tracked in separate. issues. All other items are done.

laurensvalk, 2021-07-01:
> All other items are done.

There seems to be one more case:

```python
from pybricks.hubs import ThisHub
from pybricks.tools import wait

hub = ThisHub()

print(hub.system.reset_reason())
wait(100)
hub.system.reset(0) # Already applied support#379, so here doing power-off

# The animation stays on and freezes
```

closed 2021-07-01 by laurensvalk