pybricks/pybricks-micropython #41: Implement battery charging on Prime/Inventor hubs

dlech, 2021-01-26:
- hub runtime needs to be able to shut down most things while keeping the main loop running (also see #36)
- needs to detect charger type
- needs to control charger chip
- needs to indicate status using LED
  - traditional LEGO chargers use red for charging and green for full
  - latest firmware from LEGO also uses yellow - not sure what this indicates

Vinz1911, 2021-09-12:
This is a feature which I currently really miss. Is there maybe something I can support/help here to make this work? 😊

laurensvalk, 2021-09-13:
While you're welcome to contribute directly to the driver code, there are other ways to help as well.

Perhaps the best way to help is to get other developers involved, who might contribute code or sponsor the project.

That is easier said than done, though. We're currently not as active online and on social media as we could be. Creating fun projects with Pybricks and sharing them online is usually a good start.

dlech, 2022-01-12:
The fundamentals are done now.

closed 2022-01-12 by dlech