pybricks/pybricks-micropython #43: enable buttons class on single-button hubs

dlech, 2021-01-29:
See https://github.com/pybricks/support/issues/233

If we can disable the `KeyboardInterrupt` then it makes sense to be able to use the button in the user program.

dlech, 2021-01-29:
Downside is that if we do that, then people will want to use it.

laurensvalk, 2021-01-29:
> Downside is that if we do that, then people will want to use it.

My thoughts exactly :smile:

If we do it, let's wait for v3.1 so we have a bit more time to think about it.

> enable buttons class

One consideration is that there is only one button, so using the same class might be a bit overkill. 

Maybe a single function might suffice, but this would make it different from the other hubs, and leave us less freedom to add other button features later.

JorgePe, 2021-01-29:
Funny you said that... I found recently a Raspberry Pi MIDI/sound card that uses just one button for everything (of course, if you go that route, be prepared for a whole User Interface... wanting for 3.1 or even later is a good decision)

> "When we were designing Pisound one of the main goals was to empower everyone to use Pisound and Raspberry Pi in headless mode - without an external monitor, keyboard or mouse.
> As it turns out, one button is all you need to achieve this and we call it The Button."

"https://blokas.io/pisound/"

dlech, 2021-02-26:
Thinking outside of the box:

Currently, the button raises a `KeyboardInterrupt`, but this means that to enable the button, we also lose keyboard interrupt via the REPL (and vice versa, if we want to send binary data via stdin, we also have to disable the button interrupt).

So, instead, we could expose an attribute or function to allow assigning a callback function to the button. The default would be to raise a `SystemExit` exception but the user could set it to anything they like. This would probably require enabling the scheduler in MicroPython to run the callback function, so is probably a no-go for Move hub.

laurensvalk, 2021-02-27:
That could work; it's how the official firmware does it. But if we do that, it would be nice if we consider in advance how (and *if*) we want to encourage using callback and events throughout our API, beyond just buttons. I'm not against it per se, but there's also the async flow we could have one day, and consistency would be good.

Or we could do as you suggest but partly behind the scenes, i.e. implement `Keypad.enable_stop_button(enable: bool)` which assigns or removes that callback you describe, instead of allowing a fully custom callback.