pybricks/pybricks-micropython #49: Add hub type to device info service

dlech, 2021-04-02:
We need a standard way to read the hub type via Bluetooth.

Using the BLE Device Info Service seems the simplest way to do this.

> ## 3.2 Model Number String
> The Model Number String characteristic shall representthe model number that is assigned by the device vendor.

Should we use our own model number or the `Hub No. X` from LEGO?

dlech, 2021-04-02:
We may also want to consider adding this to advertising data so that we can know which hub type this is before we even connect. This can be done with a Service Data entry. In that case, the shorter the name, the better since there is only 31 bytes times two for advertising data.

JorgePe, 2021-04-03:
> Should we use our own model number or the `Hub No. X` from LEGO?
Interesting question.
Will LEGO allow using it? It seems innocent and it's still the same Hub but lawyers this days are crazy.
And on the remote chance of using it on the same room with someone using his own Hub with original firmware, will that cause any trouble?

dlech, 2021-04-03:
AFAIK, the official LEGO firmware doesn't use the Device Information Service, so I don't think it could cause any issues like being detected by the official apps.

JorgePe, 2021-04-03:
And the opposite? Trying to connect to a Pybricks Hub and being distracted by a "pure" one? I haven't read much lately but probably not an issue.
And more than just the model number, would a major release number help? Like having Pybricks3 and Pybricks4 classes of devices (I was thinking on Boost, since you said it can't keep up with all features of the rest of the Hubs... but its not a good example since Boost itself has its own Model Number)

dlech, 2021-04-03:
> And the opposite?

See https://github.com/pybricks/support/issues/173#issuecomment-812752509 for my suggested handling of this.

> And more than just the model number, would a major release number help?

We already have separate characteristics for Firmware Version String and Software Version String (which contains Pybricks BLE service protocol version).

dlech, 2021-06-14:
done in https://github.com/pybricks/pybricks-micropython/commit/4586da20524fb28df5573505db63ab5f2326930e and https://github.com/pybricks/pybricks-micropython/commit/f38c4c0207631ae835610d175948e74b0785ca68

closed 2021-06-14 by dlech