pybricks/pybricks-micropython #51: Info about Tx Power

JorgePe, 2021-04-05:
Saw commits
https://github.com/pybricks/pybricks-micropython/commit/52bd9f2b7f7196c44bfef0822bd89ed7a15ceb30
and 
https://github.com/pybricks/pybricks-micropython/commit/8924f20cc3bcab366778c2a6d0dc95d885ee7350
(advertising Tx Power level)

Is this value the same for all Hubs? I am too tired now to check flashing several hubs but this is interesting because BLE beacons have this value factory calibrated and Apps compare it with the RSSI value to estimate a rough "proximity" range... but with 3 beacons we can use triangulation and get a better "location", I once used 3 Estimote-like beacons with EV3 for a simple proof of concept.

dlech, 2021-04-05:
I think it is 1 on SPIKE Prime/MINDSTORMS Inventor and 0 everywhere else. These are just default values. I don't have the proper equipment to do a proper "factory calibration" currently (and I don't think LEGO did this in their firmware either).

JorgePe, 2021-04-05:
Thanks.
I was afraid of something like that. Will try to make a few rough measures to see if I can get 3 hubs consistent enough for a poor-man beacon, If ever get something will report back.

laurensvalk, 2021-04-10:
It looks like this one has been answered. Feel free to open a new [discussion](https://github.com/pybricks/support/discussions) if you want to share your findings on the beacon idea.

closed 2021-04-10 by laurensvalk