pybricks/pybricks-micropython #52: Add download and run tests

laurensvalk, 2021-04-10:
[Recently added tests](https://github.com/pybricks/pybricks-micropython/tree/master/tests/pup/sensors) mainly focus on motors and sensors. In principle, it suffices to run these from only one host since the focus is on the robot side of things.

We could also use a few scripts that test the download and run process. We'd want to run these on as many devices and browsers as we want to support.

- I/O only + wait
- Long/short duration scripts
- Big/small scripts
- Tight loop print
- Raising systemexits / keyboard interrupts / other exceptions

laurensvalk, 2021-04-10:
For now, it's sufficient to run these manually from Pybricks Code.

dlech, 2021-04-21:
I've pushed a few of these that I did last week before I distracted with other things. I would still like to do some tests that create .mpy files of various sizes (e.g. 1k, 2k, 4k, etc.).

dlech, 2021-05-14:
size tests are done

closed 2021-05-14 by dlech