pybricks/pybricks-micropython #58: Add pbio function for graceful shutdown

laurensvalk, 2021-07-05:
This follows up on https://github.com/pybricks/support/issues/379 and https://github.com/pybricks/pybricks-micropython/issues/36 to implement [this todo](https://github.com/pybricks/pybricks-micropython/blob/2b06f38bd4df011c52fb53e8c30a3748ecf693d9/pybricks/common/pb_type_system.c#L49).

laurensvalk, 2021-07-05:
> We need to add a new method for proper shutdown that calls pbsys_status_set(PBIO_PYBRICKS_STATUS_SHUTDOWN);

Can we call it directly as follows or do we want a `pbio` wrapper?

```c

STATIC mp_obj_t pb_type_System_shutdown(void) {

    pbsys_status_set(PBIO_PYBRICKS_STATUS_SHUTDOWN);

    mp_hal_delay_ms(1000);

    return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_0(pb_type_System_shutdown_obj, pb_type_System_shutdown);
```
The delay is optional, but it seems nice if statements after it don't run, or should we do it in another way?

This works on all hubs, but I noticed that City Hub sometimes turns back on. I'm not sure why that is.

```python
from pybricks.hubs import ThisHub
from pybricks.parameters import Color

hub = ThisHub()

hub.system.shutdown()

# Shouldn't get here
hub.light.on(Color.RED)

```

dlech, 2021-07-05:
> The delay is optional, but it seems nice if statements after it don't run, or should we do it in another way?

I think raising `SystemExit` would make sense. 

> This works on all hubs, but I noticed that City Hub sometimes turns back on after it shutdown. I'm not sure why that is.

Watchdog timer reset?

dlech, 2021-07-05:
> Can we call it directly as follows or do we want a `pbio` wrapper?

We should use it directly.

dlech, 2021-07-05:
> I think raising `SystemExit` would make sense.

Another option is:

```C
for (;;) {
    MICROPY_VM_HOOK_LOOP
}
```

closed 2021-07-06 by laurensvalk