pybricks/pybricks-micropython #62: async/await syntax support / exposure

DietrichChristopeit, 2021-07-30:
Firstly, thank you for this great piece of work on  pybricks, it's really a gem!

I hope it's not stupid to ask for async/await.
By supporting async/await along with Tasks/Events etc. I could imagine waiting for special events to happen would be easier to implement.

I think it would be nice to have it.

dlech, 2021-07-30:
This is in our long term plan. In the mean time, have a look at https://github.com/pybricks/support/discussions/356 and https://github.com/pybricks/support/issues/357.

dlech, 2021-07-30:
And FWIW, you can already use the `aysnc` and `await` keywords. `async` doesn't really mean anything and `await` is the same as `yield from`.

DietrichChristopeit, 2021-07-30:
Oh, ok, it seems I have missed some parts... 
That's really great, thank you for pointing that out!

**BTW.:** #pybricks/support#356 and #pybricks/support#357 are a nice way of "simulating" the syntax (I know, it's rather the other way around :-) ) -- I like that way much more...

Thank you very much again

laurensvalk, 2023-01-30:
There is now a follow-up post in https://github.com/pybricks/support/issues/917, including some `async`/`await` concepts, so we can close this issue. Thanks!

closed 2023-01-30 by laurensvalk