pybricks/pybricks-micropython #95: Charging an Inventor Hub via MacBook never finishes

lobodpav, 2022-03-01:
### Environment
- MacBook Pro 16" 2019
- macOS 12.2.1
- Hub's firmware: `pybricks-micropython` commit `81d342d LICENSE: update copyright for 2022`

### Replication steps
1. Connect the MacBook to the Inventor Hub via USB-C / Micro USB cable
2. The `red` LED light indicates charging
3. **Actual behaviour**: The `red` LED never changes to `green` (even after 24 hours) and keeps on being red
4. **Expected behaviour**: The `red` shall go `green` once the battery is fully charged

When connecting the Hub to an Apple USB-C charger with the same cable, the charging led goes from `red` to `green` immediately (the hub is fully charged).

dlech, 2022-03-01:
What is the charge current with the not working charger?

    print(hub.charger.current())

lobodpav, 2022-03-01:
> What is the charge current with the not working USB port?
> 
> ```
> print(hub.charger.current())
> ```

`349`

lobodpav, 2022-03-01:
> What is the charge current with the not working USB port?
> 
> ```
> print(hub.charger.current())
> ```

Let me charge the Hub and come back.

dlech, 2022-03-01:
Actually, you should discharge the hub and come back. The charger chip in the hub will reduce the charging current as the hub nears the full point.

lobodpav, 2022-03-01:
### No charger attached
- Current: 59 mA
- Voltage: 7621 mV

### Hub connected to MacBook
- Current: 580 mA
- Voltage: 7666 mV

dlech, 2022-03-01:
Since it is ~500mA, it looks like it is detecting the port as a standard charging port and limiting charging at 500 mA. Maybe that isn't enough to ever trigger the full point? Everything other than that is controlled by the charger chip, so if it isn't getting to full, I'm not sure what else we could do about it. 

When the battery is nearly full, what is `hub.battery.current()` compared to `hub.charger.current()`? I suppose we could be getting to a point where the hub is using power as fast as we are charging.

lobodpav, 2022-03-02:
After the Hub was fully charged by Apple Charger.

### Plugged in the Apple charger
The charging led is Off.
- Charging current: 65 mA
- Battery current: 85 mA
- Battery voltage: 8257 mV

### Connected to MacBook
The charging led is RED.
- Charging current: 756 mA
- Battery current: 85 mA
- Battery voltage: 8352 mV

dlech, 2022-03-02:
> * Charging current: 756 mA
> 
> * Battery current: 85 mA
> 
> * Battery voltage: 8352 mV

So it is still charging.  The current should gradually drop and the charging should finish when it reaches 8.4V.

lobodpav, 2022-03-02:
> So it is still charging. The current should gradually drop and the charging should finish when it reaches 8.4V.

This is what I thought too. Apple charger finished charging overnight, and the Hub connected to the MacBook did not finish charging overnight even when the Hub was previously charged 😞

Will give it a second try tonight nd will report back.

dlech, 2022-03-02:
When it is this close to full, it should take less than one hour to finish.

lobodpav, 2022-03-02:
> When it is this close to full, it should take (much) less than one hour to finish.

Great, will give it a try tonight.

lobodpav, 2022-03-02:
Started charging via MacBook at 17:30 CEST with these values:
- Charging current: 452 mA
- Battery current: 89 mA
- Battery voltage: 8325 mV

30 minutes later:
- Charging current: 284 mA
- Battery current: 99 mA
- Battery voltage: 8364 mV

30 minutes later:
- Charging current: 246 mA
- Battery current: 99 mA
- Battery voltage: 8365 mV

😞

dlech, 2022-03-02:
Here is the datasheet for the battery charger if you want to read about the trickle charge phase. https://www.mouser.com/datasheet/2/277/MP2639A-1384387.pdf

I'm guessing that there is just some condition with this specific USB port that causes the chip to never get out of the trickle charge phase.

dlech, 2022-03-02:
From the data sheet, it seems like we should eventually get a timeout fault because the charge current never reaches I<sub>BF</sub> (150mA?). The timeout fault should cause CHGOK pin to oscillate which should cause the hub light to turn yellow. But it isn't clear what resets the safety timer fault.

dlech, 2022-03-02:
I suppose you could also try with the official LEGO firmware and see what happens.

lobodpav, 2022-03-02:
I have tried the official LEGO firmware now.

The charging indicator became GREEN immediatelly. The Hub reports:
- charge_current: 335 mA
- charge_voltage: 8340 mV

dlech, 2022-03-02:
I think they may have changed the meaning of red/green in recent updates, so we should be sure it actually means the same thing as Pybricks firmware before drawing conclusions.

dlech, 2022-03-02:
also, does `battery.info()` give other info on the charge state?

dlech, 2022-03-02:
Or better, just copy/paste the literal output here.

dlech, 2022-03-02:
`battery.charger_detect()` could be interesting too.

lobodpav, 2022-03-02:
> also, does `battery.info()` give other info on the charge state?

I have updated my original comment.

dlech, 2022-03-02:
so `'charger_state': 1` is `hub.battery.CHARGER_STATE_CHARGING_ONGOING` meaning that it is still charging just as in Pybricks.

dlech, 2022-03-02:
and `charger_detect` of 1 is `hub.battery.USB_CH_PORT_SDP`, which is standard downstream port, which is expected since this is a regular USB port.

lobodpav, 2022-03-02:
What the heck 😆 So it means even though they still charge the hub, they have a lower voltage threshold that is used to indicate the battery is fully charged by lighting the led GREEN. That is not bad actually, it makes sense - the user sees the hub is charged because it will probably not get charged more than it is.

dlech, 2022-03-02:
I think that on LEGO firmware, green means charging and red means done.

lobodpav, 2022-03-02:
> I think that on LEGO firmware, green means charging and red means done.

Fom what I observed in the past, it was RED when it was charging (battery was 50%). Now it is GREEN and the MS app reports 100% charge.

Will discharge the battery and come back 😊

lobodpav, 2022-03-02:
> I think that on LEGO firmware, green means charging and red means done. You could discharge the battery or charge it fully using the working charger port to be sure.

That would kinda be against common sense. RED usually means charging whilst GREEN means charged. Will test agan and confirm.

dlech, 2022-03-02:
Yeah, or maybe LEGO has seen the same problem and they use the voltage level for the indicator light now instead of the battery charger CHGOK signal.

lobodpav, 2022-03-02:
> Yeah, or maybe LEGO has seen the same problem and they use the voltage level for the indicator light now instead of the battery charger CHGOK signal.

Yep, that is what I thought. Seems like the `CHGOK` is not working properly.

lobodpav, 2022-03-02:
Discharged the battery a bit:

[image]

Connecting it to MacBook, the charging indicator goes RED as expected. Thus, GREEN means "fully" charged.

dlech, 2022-03-02:
I just had a look back at my reverse engineering notes and it does look like anything above 8.2V is considered "charging complete" in the LEGO firmware. However, 8.2V is only 75% according to the battery charger data sheet.

lobodpav, 2022-03-02:
> I just had a look back at my reverse engineering notes and it does look like anything above 8.2V is considered "charging complete" in the LEGO firmware. However, 8.2V is only 75% according to the battery charger data sheet.

How about making the indicator GREEN once the voltage reaches 8.3V?

Also wondering, does that the current charging way reduce the battery lifetime? By charging the battery indefinitely even when the voltage is 8.35 mV, that cannot be good.

dlech, 2022-03-02:
Hmm... we were hoping that the charging chip would be smart enough and we wouldn't have to worry about things like this. Back to the drawing board, I guess. :thinking: