pybricks/pybricks-micropython #96: chore: update to GCC 11

dlech, 2022-03-06:
Attempted this with 11.2-2022.02, in https://github.com/pybricks/pybricks-micropython/tree/gcc-11 however, it is buggy and gives

   internal compiler error: Illegal instruction

on some floating point code. 

https://bugs.linaro.org/show_bug.cgi?id=5825

laurensvalk, 2022-09-26:
Everything seems to be working fine in `12.2.MPACBTI-Bet1`, so maybe we can wait until it is officially released and skip GCC 11.

Move Hub build size is bigger by about 500 bytes.

dlech, 2022-09-26:
Another alternative it to just keep using 10.x for the time being since it ships with Ubuntu 22.04 and then update to whatever ships with 24.04 in a few years.

laurensvalk, 2022-10-04:
We will stick with GCC 10, so we can close this item.

> Another alternative it to just keep using 10.x for the time being since it ships with Ubuntu 22.04 and then update to whatever ships with 24.04 in a few years.

That's a good idea. Opened issue for it: https://github.com/pybricks/support/issues/733

closed 2022-10-04 by laurensvalk