pybricks/pybricks-micropython #103: chore: update to MicroPython 1.19

dlech, 2022-06-14:
MicroPython 1.19 hasn't been released yet, but will be soon. There are more than the usual update tasks with the version (mostly due to new mpy ABI version), so making note of them here:

- [ ] Publish `mpy-cross-v6` Javascript package.
- [ ] Publish `mpy-cross-v6` Python package.
- [ ] Make use of `MP_REGISTER_MODULE` macro.
- [ ] Build and use mboot on SPIKE hubs.

laurensvalk, 2022-06-23:
The rebase step itself seems to have gone without a hitch: https://github.com/pybricks/micropython/tree/pybricks-v3.2

> Make use of MP_REGISTER_MODULE macro.

So the only commit I had to skip was:

https://github.com/pybricks/micropython/commit/bf835f45e56c381b2ae9134a8414c84315dd64dd

I mainly just did the rebase to ensure that <*py/builtinimport.c: Support builtin-package as nested modules.*> still rebases without conflicts, and it does. I'll check in on our next call before proceeding.

laurensvalk, 2022-07-01:
Perhaps not technically related to this upgrade, but I've added one item above nonetheless. Maybe we can do it while we're at it.

It would nice if we can import the same modules on the virtualhub as on the real hubs. Currently this gives an import error:

```python
from umath import sin, pi
```

dlech, 2022-07-01:
> ```python
> from umath import sin, pi
> ```

:wink: https://github.com/micropython/micropython/pull/8844

closed 2022-08-12 by dlech