pybricks/pybricks-micropython #125: reboot in update mode no longer works on hubs with BLE bootloader

dlech, 2022-11-11:
Probably an issue with LTO, although we already added the `used` attribute in https://github.com/pybricks/pybricks-micropython/commit/b84ac3d91cbf6219f8b2431033e5a1a693ea090d to fix this before. Not sure what else could have changed in the meantime.

laurensvalk, 2022-11-11:
> Not sure what else could have changed in the meantime.

We added the save step on power off.

So perhaps try it without running a newly downloaded program as well.

dlech, 2022-11-11:
This reset skips any "shutdown" and calls `NVIC_SystemReset()` immediately.

laurensvalk, 2022-11-11:
Maybe something with the no-init RAM section we added to the linker script?

dlech, 2022-11-11:
Actually, I think this is problem with part of `pybricksdev` not being updated to the new download and run.