pybricks/pybricks-micropython #139: Compilation error: implicit declaration of function 'mp_hal_stdout_tx_flush'

pbochynski, 2023-01-17:
I am not able to build a firmware from sources after this commit:
https://github.com/pybricks/pybricks-micropython/commit/d7b04e7ecbf1093994ee23fc763d41eeb988d942

I get the error:
```
CC ../../bricks/_common/micropython.c
../../bricks/_common/micropython.c: In function 'pbsys_main_run_program':
../../bricks/_common/micropython.c:296:5: error: implicit declaration of function 'mp_hal_stdout_tx_flush'; did you mean 'mp_hal_stdout_tx_strn'? [-Werror=implicit-function-declaration]
  296 |     mp_hal_stdout_tx_flush();
      |     ^~~~~~~~~~~~~~~~~~~~~~
      |     mp_hal_stdout_tx_strn
cc1: all warnings being treated as errors
```
I use command:
```
make technichub
```

It works with the previous commits. 
How can I fix it?

laurensvalk, 2023-01-17:
Have you tried cleaning the build with `make clean-technichub` ?

pbochynski, 2023-01-17:
Yes - that was the first thing I tried.

pbochynski, 2023-01-17:
Thanks - that was it! I never worked with submodules and didn't know about this extra step.

closed 2023-01-17 by pbochynski

laurensvalk, 2023-01-17:
> [d7b04e7](https://github.com/pybricks/pybricks-micropython/commit/d7b04e7ecbf1093994ee23fc763d41eeb988d942)

In case you run into this next time: In the commit details you can see that the MicroPython submodule was updated:

[image]