pybricks/pybricks-micropython #189: ImportError: can't import name multitask

sgatu, 2023-07-21:
I'm not sure what I'm doing wrong. 

I have a Prime Hub and installed last firmware version, 3.2.3. The thing is that when I upload my code to the hub though code.pybricks.com I get the next error.

`ImportError: can't import name multitask`

on this line

`from pybricks.tools import wait, multitask, run_task, Matrix`

Am I missing something? Is not enough with the the firmware?

closed 2023-07-21 by sgatu

laurensvalk, 2023-07-21:
These features are still in beta. To try them out, please use https://beta.pybricks.com/ and install the firmware from there.

Thanks!

sgatu, 2023-07-21:
Yes, found out by looking into the support page, sorry for the issue.

laurensvalk, 2023-07-21:
No problem, I'm glad this is getting some testing (despite the lack of documentation!)

Please let us know how it goes.

The benefit of being in beta, is that we can still easily change these features as it won't affect people on the stable version.