pybricks/pybricks-micropython #220: Axes for gyro

MonongahelaCryptidCooperative, 2023-12-08:
Something is off about the axes for gyro use in DriveBase/Hub setup.  Our robot has a vertically oriented hub, facing the rear and per the documentation should have the following settings;

Top_side: + Axis.Z
Front_side - Axis.X

This does not work and results in the robot spinning indefinitely. 

However it works perfectly with the following settings:
Top_side: + Axis.Y
Front_side:+ Axis.Z

Indeed it was the above settings that performed flawlessly at the state competition.

MonongahelaCryptidCooperative, 2023-12-08:
I also made a bot with the charging port facing forward the the screen up. Per documenation this this be top +X, front_side + Z. In reality front + Z, Top + Y is what works.

dlech, 2023-12-08:
https://docs.pybricks.com/en/latest/signaltypes.html#reference-frames

[image]

> Our robot has a vertically oriented hub, facing the rear and per the documentation should have the following settings;
> 
> Top_side: + Axis.Z Front_side - Axis.X

Assuming you mean like the image above, it says `front_side=Axis.X, top_side=-Axis.X` rather than either of the configurations you have written here.

MonongahelaCryptidCooperative, 2023-12-08:
actually I thik I was just mistaken. That actually works. Ignore what I wrote before.

MonongahelaCryptidCooperative, 2023-12-08:
We just made a dumb mistake, everything is fine.

closed 2023-12-08 by MonongahelaCryptidCooperative