pybricks/pybricks-micropython #228: run_until_stalled() stops working when adding args.

antonvh, 2024-01-14:
When I add a duty limit, the command starts running the motor in the wrong direction and fails to execute run_target() later in the program. Here's the code I used. See comments for expected and unexpected results.

``` python
from pybricks.pupdevices import Motor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.tools import wait, StopWatch

steer = Motor(Port.A, Direction.CLOCKWISE)
# This works. Motor starts Clockwise, 100 deg/s
steer.run_until_stalled(100)
print("stalled at: ", steer.angle()) # returns
steer.reset_angle(300)
print("reset angle at: ", steer.angle()) # returns 200, neatly
steer.run_target(100, 0)
# Motor runs as expected.
print("Ran to: ", steer.angle()) 
steer.close()

# Now the trouble begins.
steer = Motor(Port.A, Direction.CLOCKWISE)
# Expecting the motor to run clockwise, because 100>0. But motor starts CCW
steer.run_until_stalled(100, Stop.COAST, 10) # < the only difference are the params here
print("stalled at: ", steer.angle()) # returns
steer.reset_angle(200)
print("reset angle at: ", steer.angle()) # returns 200, neatly
steer.run_target(100, 0)
# returns a number between -5 and 7 BUT the motor did NOT move!
print("Ran to: ", steer.angle()) 
```

laurensvalk, 2024-01-14:
Does this happen when you don't use `close()` too?

laurensvalk, 2024-01-14:
Can you please open new issues here? https://github.com/pybricks/support I keep moving them :smile: